TheClams/SystemVerilog #1: Only index files based on a .f filelist

DeeeeLAN, 2019-09-26:
Not sure if you want tickets here or not. I will move it over to bitbucket if you would prefer to keep them there for now. 

One nice thing eclipse does is it only indexes files for autocomplete based on a file list. This way, I won't get autocomplete functions from code that I won't actually be compiling. Since basically every SV project already uses .f files (and if not, it is trivial to add one specifically for indexing purposes), this would be a nice thing to have. 

For extra credit, you could add a badge to the files that are indexed, so you know sublime is picking them up.

TheClams, 2019-09-27:
Indexing is handled by sublime itself, not this plugin.

DeeeeLAN, 2019-09-28:
How about a function that parses .f files and appends to the "index_include_paths" list in the project settings. You would have to re-run it any time the file list changes, but that could work, I think.

TheClams, 2019-09-30:
Sure, but that is not specific to Verilog, so for me it is outside of the scope of the plugin, it can be done in another plugin with no dependency on this plugin.

closed 2019-09-30 by TheClams