TheClams/SystemVerilog #6: NavBar fails if there is an instance within a block comment

markusdd, 2020-02-23:
I observe the funny effect that on my outermost toplevel file, the navbar refuses to build and just prints a few '+' signs. One level below, where complexity is just very slightly less, everything shows up perfectly.

This is how it looks:
[image]

The cause is an instance on the toplevel which does not exist for the current project and sits within a block comment. In the console I get a message von NavBar saying 'Can't parse XYZ' (XYZ being the module declaration within the block comment which is inactive)

So I guess the parser needs to be adapted to ignore module statements within a block comment.

markusdd, 2020-02-23:
Had a quick look and I believe the bug is in this function (from verilog_naviagtion.py):

def getObjList(view):
    r = view.sel()[0]
    p = r'(?s)^[ \t]*(?:virtual\s+)?(class|module|interface)\s+(\w+\b)'
    nameList = []
    view.find_all(p,0,r'\1 \2',nameList)
    # print(nameList)
    return nameList

I believe this does return the commented module as a valid object, which it shouldn't.
So basically this needs to be some weird look-back regex which makes sure if there was a /* there was also a */, otherwise we are within a block comment or something like that...

TheClams, 2020-02-23:
Thanks for the report and the first investigation. 
This is fixed in the dev branch.

closed 2020-02-23 by TheClams

markusdd, 2020-02-23:
Using the comment scope as filter is a very neat fix. Thanks for reacting so very quickly!