TheClams/SystemVerilog #12: Adding comments to Outline

ldm1417, 2020-04-25:
Hello,
I've figured out that SystemVerilog package works with Outline package and is able to add there labels of the **begin** and **fork** blocks of code.
Is it possible to add comments to the Outline? Let's say all the comments starting with //- (double slashes and single dash) or //-- (double slashes and double dashes)?
I tried to do this with the PackageResourceViewer package and BetterGoTo package, but things do not work perfectly. Probably I'm missing something ... Since you created the package and more familiar with it, probably you could cap the issue more... 
Could you please help with adding the tagged comments (//- and //--) to Outline?
Thank you a lot in advance!

ldmas, 2020-04-25:
Please refer to [https://github.com/warmdev/SublimeOutline/issues/2#issuecomment-256441587 ](url)

ldm1417, 2020-04-27:
Any response?

closed 2020-04-30 by TheClams