TheClams/SystemVerilog #14: How to modify declaration syntax for navigation?

Doreldev55, 2020-05-08:
This is not an issue but a guidance request please.
I am using your extension for a long time and it's awesome. I need to modify two things for a slightly different syntax capture! maybe you can give me a hint I was struggle to find the exact location.
1. I need signal declaration to capture also signals with circular parentheses as: 
wire (struct_name) signal_name;
2. I need that module decleration navigator will search for module name only by files name as:
test.sv contain module declaration without mention "test" but when I create an instance I use as usual
test i_test(
Bla);

Thank you

TheClams, 2020-05-08:
I don't understand at all your question:
- What do you mean by "signal declaration to capture something" ? you are
talking highlightnig or something else ?
- What is "module declaration navigator" ?

Doreldev55, 2020-05-09:
Thank you for the fast answer and for your time.
About the **signal** declaration:
I added a screenshot that shows that the signal declaration is identified only for the syntax mentioned in the second line. 
[image]
Where is the regex that responsible for signal declaration? so I will be able to add the first line syntax.

About the **module** declaration:
It's harder to explain but I will try one more time.
When I hover on an instance I would like the tool to also search for the module by the file name and not only by the body declaration in each file.
(for this example the declaration "module  module_name();" is not written in the body of the file "prt_tx_datapath")
[image]

TheClams, 2020-05-09:
The regexp for signal declaration is `re_decl` in verilogutil.py
You';; have to be careful about the capture group since they are sued to
break-down the different properties of the signal (typically I guess you'll
need to use `(?:...)` to avoid changing the capture group index

The module declaration is not the result of a search in all files, but
simply using the indexing done by Sublime Text: the syntax assign the scope
`entity.name` when a module is declared, and then the definition is found
using `lookup_symbol_in_index`. This is done in the `get_type` function in
verilog_navigation

Doreldev55, 2020-05-09:
Thanks a lot, it works perfectly!

closed 2020-06-27 by TheClams