TheClams/SystemVerilog #17: tooltip conflict with sublime default function.

xgzi, 2020-07-31:
when i haven't disabled tooltips( by default) . i find there are some errors, which caused my sublime stuck.
when i disabled it, problem fixed.

```
	"sv.tooltip" : false, // True to display information in a tooltip (sublime 3072+), false to use the status bar instead
	"sv.tooltip_hide_on_move" : false, // True to hide the tooltip when the mouse move away from the tooltip
	"sv.tooltip_show_module_on_port" : false, // True to include module type and instance when showing tooltip for a port
	"sv.tooltip_show_refs" : false, // True to show reference to module/interface/...
	"sv.tooltip_show_signal_links" : false, // True to show links to driver/reference for signals
```

TheClams, 2020-08-07:
You need to be more precise about the issue and provide a way to reproduce it, otherwise I cannot help.

awall13, 2020-08-30:
This is a fun one. I'm really frequently getting crashes when I mouse over certain signals. I narrowed it down, such that the following SV, on its own, will reliably crash Sublime when I mouse over or run the "verilog type" command on BAR_A/B/C/D. Interestingly, reducing the indent before the commas in foo_t, or moving the commas onto the prior lines, resolves the crash, as does reducing the number of enumerations in foo_t, moving the list of enumerations for foo_t onto a single line, or removing the range of enum declarations in foo_t. 

`
typedef enum logic [4:0]
    { FOO_A[1]
    , FOO_B[1]
    , FOO_C[1]
    , FOO_D[1]
    , FOO_E[1]
    , FOO_F[1]
    , FOO_G[1]
    , FOO_H[1]
    , FOO_I[1]
    , FOO_J[1]
    , FOO_K[1]
    } foo_t;

typedef enum logic [2:0] {BAR_A, BAR_B, BAR_C, BAR_D} bar_t;

`

TheClams, 2020-08-30:
Thanks for taking the time to create a simple reproducible example: funny thing is it actually does not even crash on my machine, but it looked a bit slower than normal, and I finally pinpoint that this seems to be due to a regexp running out of stack.
I'll need some time to actually find a proper fix, but I'm on it.

TheClams, 2020-09-03:
I tested the fix this week, seems to work fine: tell me if it is fine on your side as well

awall13, 2020-09-03:
Thanks! Looks like it's resolved for me as well.

closed 2020-09-04 by TheClams