TheClams/SystemVerilog #21: Support for module instance parameters in macro

awall13, 2020-10-12:
Is it possible to add support for module instance parameters to be defined in a macro? e.g. 

```
module_name `PARAM_DEF u_inst3 (.*);
```

Currently u_inst3 has a scope of:
```
source.systemverilog support.function.generic.systemverilog
```

Adding the text "(`\w+\b)?\s*" in the module-inst section of the syntax definition should allow this to be recognized (along with renumbering the captures below it). 
```
- match: '\b({{id}})\s+(?!intersect\b|and\b|or\b|throughout\b|within\b)(`\w+\b)?\s*({{id}})\s*((?:\[(\d+)(\s*(\:)\s*(\d+))?\])|(?:\[([^\]]+)\]))?\s*(\(|$)'
```

Thanks!

awall13, 2020-10-13:
Also, to handle the same thing for the "show hierarchy" and "show navbar" commands, where a module declaration or instantiation has parameters hidden in a macro, there are 3 lines that can be edited in verilogutil.py, from what I can see. 

At the top:
```
re_inst  = r'^\s*(virtual)?(\s*)()(\w+)\s*(#\s*\([^;]+\)|`\w+)?\s*()\b'
```
And the 2nd and 12th lines of parse_module
```
    re_str = r"(?s)(?P<type>module|interface)\s+(?P<name>"+mname+r")(?P<import>\s+import\s+.*?;)?\s*(`\w+)?\s*(#\s*\((?P<param>.*?)\))?\s*(\((?P<port>.*?)\))?\s*;(?P<content>.*?)(?P<ending>endmodule|endinterface)"
```
```
        re_str  = r'^[ \t]*(\w+)\s*(?:#\s*\([^;]+\)|`\w+)?\s*\b(\w+)\b(?:\s*\[[^=\^\&\|,;]*?\]\s*)?\s*\('
```

This would address these two cases
```
module_name `PARAM_DEF u_inst3 (.*);
```
and
```
module module_name `PARAM_DEF (
```

TheClams, 2021-03-11:
The main issue is that a macro can literally be anything, I have seen some legacy code where it is used to do an assign with delay: not knowing the content of the macro makes it impossible to know the difference between your case and this example. So basically trying to handle this case is more likely to break other stuff, so I prefer to keep as is,.

closed 2021-03-11 by TheClams