TheClams/SystemVerilog #23: code gets deleted when i align or reindent

shafqatu, 2020-10-27:
When import package and have parameterized module at the sometime then alignment and re-indent does not work, they destroy the code. 

I am also attaching the screenshoots.

[image]
[image]
[image]

TheClams, 2020-10-28:
You are missing a comma after the rst_n which might explain the problem.
And next time please provide raw text instead of screenshot it makes
my life easier when i have to reproduce the problem.

shafqatu, 2020-10-28:
`module test_sram
	import test_pkg::*;
#(
	parameter LENGTH      = 8192,
	parameter IS_IMEM     = 1   , // this is just for simulation to lead hex files
	parameter MEM_TYPE    = 0   , // 0 = single cycle pipeline, 1 = multi cycle
	parameter NUM_OS_RW   = 8   , // number of outstanding read and writes, beofe things get blocked
	parameter ALMOST_FULL = 2   , // this define how long before busy signals is set high
	// this is can set depending on pipes between blocks, should be less than NUM_OS_RW
) (
	input                clk    , // Clock
	input                rst_n  , // Asynchronous reset active low
	input  abee_master_t ib_sram,
	output abee_slave_t  ob_sram
);

endmodule`

shafqatu, 2020-10-28:
module test_sram
	import test_pkg::*;
#(
	parameter LENGTH      = 8192,
	parameter IS_IMEM     = 1   , // this is just for simulation to lead hex files
	parameter MEM_TYPE    = 0   , // 0 = single cycle pipeline, 1 = multi cycle
	parameter NUM_OS_RW   = 8   , // number of outstanding read and writes, beofe things get blocked
	parameter ALMOST_FULL = 2   , // this define how long before busy signals is set high
	// this is can set depending on pipes between blocks, should be less than NUM_OS_RW
) (
	input                clk    , // Clock
	input                rst_n  , // Asynchronous reset active low
	input  abee_master_t ib_sram,
	output abee_slave_t  ob_sram
);

endmodule

shafqatu, 2020-10-28:
`module test_sram
	import test_pkg::*;
#(
	parameter LENGTH      = 8192,
	parameter IS_IMEM     = 1   ,
	parameter MEM_TYPE    = 0   ,
	parameter NUM_OS_RW   = 8   ,
	parameter ALMOST_FULL = 2
) (
	input                clk    , // Clock
	input                rst_n  , // Asynchronous reset active low
	input  abee_master_t ib_sram,
	output abee_slave_t  ob_sram
);

endmodule`

shafqatu, 2020-10-28:
module test_sram
	import test_pkg::*;
#(
	parameter LENGTH      = 8192,
	parameter IS_IMEM     = 1   ,
	parameter MEM_TYPE    = 0   ,
	parameter NUM_OS_RW   = 8   ,
	parameter ALMOST_FULL = 2
) (
	input                clk    , // Clock
	input                rst_n  , // Asynchronous reset active low
	input  abee_master_t ib_sram,
	output abee_slave_t  ob_sram
);

endmodule

shafqatu, 2020-10-28:
I hope this helps

TheClams, 2020-10-29:
It works fine for me:
[image]

TheClams, 2021-03-11:
Finally managed to re-create the issue : it happens only if you run the alignement outside of the parameter/port list ...
Anyway this is now fixed in 3.0.6

commit 2250b41d (2021-03-11): [Alignment] Fix issue with module using import and parameter

Beautify text on a whole file with a parametrized module using import
 statement after the module name would completely scrap the file ...

Also try to alter slightly the hot reload to try avoiding error message
(the loading works but still spam the console with some errors ...

Close #23
closed 2021-03-11 by TheClams