TheClams/SystemVerilog #26: Reindenting typedefs indents poorly

DeeeeLAN, 2021-02-16:
Code like:
```
typedef class Class1;
typedef class Class2;
typedef class Class3;
``` 
will indent to:
```
typedef class Class1;
  typedef class Class2;
    typedef class Class3;
```

DeeeeLAN, 2021-02-17:
Also, jumping to hierarchy with a typedef just takes you to the typedef instead of the actual class.

TheClams, 2021-02-21:
The alignment is fixed in the next version of the plugin.
For your second remark, I'll need more details to reproduce it

DeeeeLAN, 2021-02-22:
I mean if I have the cursor on a class object and I want to jump to its definition, it will jump to the line where the typedef is declared instead of to the actual class definition. Not sure if that is the plug-in or ST feature though come to think of it.