TheClams/SystemVerilog #28: cannot turn off autocomplete

aaleclaire, 2021-03-12:
Hi there!

I am not sure if this is an issue with the systemverilog package, or the basic sublime package.  I have tried to turn off autocomplete in both the systemverilog package and the general sublime preferences.   I have tried to use the sublime forum to resolve the issue, but without success.  I have also tried to uninstall and re-install sublime, but I am suspicious that windows did not delete all the sublime files during the uninstall.  I say that because after I reinstalled sublime, the "autocomplete false" settings were still there.

I am suspicious that this issue might be with sublime and not the SV package.  I say that because if I type always_ff and hit tab, the template for the always block does not appear.  

I am operating on my own dime, but I am pretty motivated to get this resolved.  If I can pay for support, I will...  Any advice is appreciated.

TheClams, 2021-03-12:
Something is not entirely clear for me: you are saying that the autocompletion for always block is not present, so autocompletion is correctly off since you set the sv.disable_completion to true ? which is what you wanted if I understood correctly.
Is you issue that you still have basic autocompletion from Sublime ? For sublime there are two settings:
 - ```"auto_complete" : false``` avoids autocompletion to appear while typing
 - ```"tab_completion" : false``` avoids also must be set to off so that when you hit tab the auto-completion popup does not appear

If you still get auto-complete what are you dong exactly which makes the auto-complete pop-up ?

aaleclaire, 2021-03-12:
I was mistaken about auto_complete; it was indeed turned off.  It was
tab_complete that was causing me grief.  I didn't know that tab_complete
existed until last night, and it is turned off now.  I appreciate your
help, thank you!

Aaron

closed 2021-03-12 by TheClams