TheClams/SystemVerilog #32: Instance generator doesn't work with package declaration defined in module port list.

mm4dd, 2021-06-29:
Hello,

Firstly I wanted to say how excellent this plugin is and how helpful it has been for my systemverilog coding.

I do have one request when you are doing an update to the plugin. When trying to use a create instance from a module that imports a package in the port list like below the module instance is not created.  A normal module port list works fine. 

`module example import RTL_pkg::*;
               #(parameter                                    REV_CORE = 16'h0000,
                 parameter                                    NUM_CORE  = 16'h0000)
                (input  logic                                 clk_sys,    
                 input  logic                                 reset_n,    
                 input  logic [(ADDR-1):0]                    address,    
                 input  logic                                 write       
                ); `

TheClams, 2021-07-04:
I only get an issue if I ommit the endmodule like in your example. But if I add the endmodule then it works as expected

mm4dd, 2021-07-04:
Thanks for having a look at it TheClams. Unfortunately, it never works for me with the package defined within the module port list. For example this example works without the package defined in the port list.
`module reg_data #(parameter WIDTH = 1)
                 (
                 	input  logic                   clk,
                 	input  logic                   reset_n,
                 	input  logic [(WIDTH-1):0]     data_in, 
                 	output logic [(WIDTH-1):0]     data_out
                 );

  // Register data
  always_ff @(negedge clk or negedge reset_n)
  begin
    if (!reset_n)
    begin
      data_out  <= 'b 0;
    end 
    else 
    begin
      data_out  <= data_in;
	 end	
  end 

endmodule : reg_data`  

However this example does not work with the package defined in the port list.
`module reg_data import RTL_pkg::*;  

                 (
                 	input  logic                   clk,
                 	input  logic                   reset_n,
                 	input  logic [(WIDTH-1):0]     data_in, 
                 	output logic [(WIDTH-1):0]     data_out
                 );

  // Register data
  always_ff @(negedge clk or negedge reset_n)
  begin
    if (!reset_n)
    begin
      data_out  <= 'b 0;
    end 
    else 
    begin
      data_out  <= data_in;
	 end	
  end 

endmodule : reg_data`. 

Could you possibly try this second example? 
Thanks
mm4dd

TheClams, 2021-07-04:
I tried both ST3 and ST4, but I still can't reproduce:
[image]

Maybe check which version of the plugin you have ?

mm4dd, 2021-07-04:
Hi TheClams. Thanks for trying my code. I have uninstalled and reinstalled the pluggin and it is V2.24.1. I think it is the latest. I am not sure what the issue is but it has never worked properly for me using packages with my setup. I appreciate taking the time to look at my issue anyways. Cheers mm4dd.

TheClams, 2021-07-04:
Ok this is indeed the latest version for ST3...
There is explicit code to handle the import statement so it is quite strange it does not work for you.

Which OS are you using ? I tested this only on Windows, I could test it again on Linux during the week.

mm4dd, 2021-07-05:
I finally have managed to get the instance tool to work for my package instance port list. I had to reinstall the package a few times to get it to finally work. Thanks for all you assistance TheClams.

closed 2021-07-05 by mm4dd