TheClams/SystemVerilog #43: Lof of CPU usage when I type dot

mickaelgnb, 2021-12-20:
When I type a dot ( "."' ) after an object, my sublime text freeze during few seconds.

During this freeze, I have a big usage of CPU with plugin_host process.

I am using sublime text 4 - 4121 and for SV plugin 3.1.2

mickaelgnb, 2022-01-05:
Hello,

Here my code (quite simple)  :  

````systemverilog
class tb_param extends uvm_component;

    rand int nb_packet;
    rand int latency_flit_min;
    rand int latency_flit_range;
    rand int latency_packet_min;
    rand int latency_packet_range;
    rand int payload_size_min;
    rand int payload_size_max;
    rand int hit_rate;
    rand int nb_lanes_used;
    int latency_flit_max;
    int latency_packet_max;
    int nb_parsers_enable=8;

    constraint c_nb_packet  {
        nb_packet dist {
            2500 := 75,
            3000 := 25
        };
    }

    constraint c_hit_rate {
        hit_rate dist {
            0 :/ 25,
            [1:99] :/ 25,
            100 :/ 50
        };
    }

    constraint c_lantency {
        latency_flit_min inside {0, 10, 50};
        latency_flit_range inside {0, 10, 50};
        latency_packet_min inside {0, 10, 50};
        latency_packet_range inside {0, 10, 50};
    }

    constraint c_payload_size {
        payload_size_min >= 0;
        payload_size_min < 2000;
        payload_size_max >= payload_size_min;
    }

    constraint c_nb_lanes_used {
        nb_lanes_used inside {1,2,4};
    }

    `uvm_component_utils_begin(tb_param)
        `uvm_field_int(nb_packet, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(payload_size_min, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(payload_size_max, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(latency_flit_min, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(latency_flit_max, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(latency_packet_min, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(latency_packet_max, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(hit_rate, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(nb_parsers_enable, UVM_ALL_ON + UVM_DEC)
        `uvm_field_int(nb_lanes_used, UVM_ALL_ON + UVM_DEC)
    `uvm_component_utils_end

    function new(string name, uvm_component parent);
        super.new(name, parent);
    endfunction : new

    function void post_randomize();
        latency_flit_max = latency_flit_min + latency_flit_range;
        latency_packet_max = latency_packet_min + latency_packet_range;
    endfunction : post_randomize

endclass : tb_param

````

TheClams, 2022-01-06:
Thanks for the example code.
What are you doing to trigger the issue ? I tried creating a variable `tb_param tb;` and typing `tb.` : i see no big CPU usage and immediately get the completion with all the members.

mickaelgnb, 2022-01-07:
Is there a debug mode to see what's going on?

TheClams, 2022-01-07:
In preferences->packages settings->SystemVerilog->settings you can set sv.debug to true and then it should print some information in the console (view->show console).
I'll check later if there is enough debug print around completion and if not I'll add some.

mickaelgnb, 2022-02-24:
I localized the issue, it comes from when there are lot of lines in the current file and it is due to this following regexp :

https://github.com/TheClams/SystemVerilog/blob/7eca705e87f87b94478fe222fc91d54d488cc8e3/verilogutil/verilogutil.py#L96

I attached an example:
[test.tar.gz](https://github.com/TheClams/SystemVerilog/files/8134076/test.tar.gz)

to reproduce the problem, type "o." inside new function like this :

```systemverilog
class my_class2 extends my_class;
	function new();
        o.
    endfunction : new
endclass
```

TheClams, 2022-02-24:
Thanks for that i'll have a look this week end

TheClams, 2022-02-26:
Thanks again I was able to reproduce the problem and the fix should be robust: tell me if you still have similar problem.

mickaelgnb, 2022-03-07:
WIth the 3.1.5, I have no more this issue, thanks for your fix !!

closed 2022-03-07 by mickaelgnb