TheClams/SystemVerilog #50: auto completion of module parameter not working for tag 3.1.5 in ST4

PeteTreeUk, 2022-03-31:
In commit of July 21, 2021 ((auto completion of module parameter not working for tag 3.0.9 in ST4 #35) this issue was fixed, but now the issues its back again.

The problem is:
With sublime_text_build_4126_x64 , when I update the packages I got SystemVerilog 3.1.5.
When I instantiate a module that has parameters I can still see the paraments hoovering the mouse, but the autocompletion of the parameters no logger works. also the auto completion does not work for the inputs and outputs.

All of this works well in the ST3 that I have installed with the same settings.

TheClams, 2022-04-01:
Can you provide an example, because on the parameterized module I used I do not see the issue

PeteTreeUk, 2022-04-01:
In my example the issues happens when vset_load.vams calls ams_vres.vams

I have a put a simplifications of my files, and my project... and aso the user systemverilog settings
this same setup works well in ST3 but not in ST4.... some months ago it did work, but now with the latest updates it doenst.

[example.zip](https://github.com/TheClams/SystemVerilog/files/8400158/example.zip)
s

TheClams, 2022-04-02:
Thanks, it will be fixed in next version

PeteTreeUk, 2022-07-25:
> 

Hi, there was a few releases since this issue, but because the issue still persists, I'm still using ST3. 
Do you have any plans or date to release this soon? I would like to start using ST4. Thank you.

TheClams, 2022-07-25:
I opened your example, and it works fine for me.

PeteTreeUk, 2022-07-27:
I had to re install ST4 from scratch, and only install the systemVerilog plugin, to make sure my setup is correct.

what I see now is that if type:
ams_vres #(.)      ----- like this , Sublime does not show me the parameter options when i put the . inside the #()
ams_vres #(.) instance() ----- in this situation where I have  declared the name of the instance if i go back in the line then #(.) gives the options of autocomplete of the parameters.

In ST3 there is no need to declare the instance name for the module parameters to show the autocomplete options while in ST4 you need to add the instance name.

Please let me know if you see the same behaviour.
Could this be fixed to resemble the ST3 behaviour ?

TheClams, 2022-07-27:
I still cannot reproduce your issue: 
 - I open the folder with all the files from your example
 - open a new file, start to type ams_vres #() and when I type the . between the parenthesis I still get the auto-completion. Also works in the vset_loads.vams when I remove the instance name, I still get autocompletion for the parameters.
[image]

PeteTreeUk, 2022-07-28:
Your example works with me too, thank you.
I know the reason i see the bug, its because i normally don't do much conde from scratch, and instead add code.

Here is the Scenario that it dos not work, where there is the bug:

type in line 2:    ams_vres #(.ttol(10n)) i_vres1 (.p(p), .n(n));  // the autocomplete for parameters work
now go above and type in line 1: ams_vres #()  //the auto complete does not work for parameters unless you have typed the instance name, in which case the pins autocomplete does not work either, unless you put he semicolon at the end and then go back and try the auto complete, then it will work!

TheClams, 2022-07-29:
Thanks, I could reproduce it and it is now fixed

closed 2023-04-09 by TheClams