TheClams/SystemVerilog #52: How to set a project?

slumpedyeti, 2022-06-30:
I'm just getting started with the tool. Great work, by the way!

The [documentation page](https://sv-doc.readthedocs.io/en/latest/) mentions, "Note that advanced features, like completion or tooltips, are not available if no project is defined." I'm wondering where the project can be specified.

I currently have the folder project open on Sublime to see the entire hierarchy on the left navigation pane. I wonder if linking the project to the SystemVerilog package will allow me to select a module instance's port in File A and navigate (GoTo Driver) to File B's module port declaration line. 

Thanks!

TheClams, 2022-06-30:
The dcumentation is only refering to the sublime project: if you have a sublime project then everything should work

slumpedyeti, 2022-06-30:
Not sure why it's not working then. I'm going to try and reset the project and maybe even reinstall the package. Thanks!

TheClams, 2022-07-02:
If you can give me a reproducible test showing your issue, maybe I can have a look

closed 2023-04-09 by TheClams