TheClams/SystemVerilog #56: File encode

godenfreemans, 2022-10-04:
About file encode, may lead some error.
in verilogutil.py
```
@functools.lru_cache(maxsize=32)
def get_type_info_file_cache(fname, var_name, fdate):
    with open(fname) as f:
        flines = f.read()
        ti = get_type_info(flines, var_name)
    return ti
```

Maybe make a check before open? like:
```
    with open(fname, 'r', encoding=encoding) as f:
        flines = f.read()
```

rayqqqm, 2023-01-14:
Also meet the similar problem.

system : Windows 10 in Simplified Chinese language. The default encoding of the system is GBK.
sublime: 4143
systemverilog: st4-3.1.11

When I deal with some UTF-8 encoded verilog file with Chinese characters，this plugin is basically crashed.

for example:
1.
When I try to use "Verilog:Instantiate Module"，nothing happens，and the console shows:

Traceback (most recent call last):
  File "C:\GreenSW\SublimeText\Data\Installed Packages\SystemVerilog.sublime-package\verilog_module.py", line 434, in on_select_file_done
UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 388: illegal multibyte sequence.

seems like the 

```
                with open(fname, "r") as f:
                    flines = str(f.read())
```
doesn't process the encoding properly.

2. 
When I try to use "Verilog:Find Instance"，the console shows:

Traceback (most recent call last):
  File "C:\GreenSW\SublimeText\Data\Installed Packages\SystemVerilog.sublime-package\verilog_navigation.py", line 1670, in <lambda>
  File "C:\GreenSW\SublimeText\Data\Installed Packages\SystemVerilog.sublime-package\verilog_navigation.py", line 1682, in findInstance
UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 388: illegal multibyte sequence.

seems like the 

```
            with open(fn) as f:
                txt = f.read()
```
doesn't process the encoding properly.

Also there are lots of similar ”UnicodeDecodeError“ problems in "verilog_completion.py" "verilogutil.py" and so on.

I'm not a python coder but I guess maybe the python dosen't process the encoding right when the file encoding is different from the default system encoding?

TheClams, 2023-01-16:
Duplicate of #60 : fixed in last version

closed 2023-01-16 by TheClams