TheClams/SystemVerilog #58: Syntax highlighting inconsistent for port connections of sub module

maxbjurling, 2022-12-13:
Truly a minor issue, but still:  the color of the dot (".") in port connections of a sub module is different if using implicit named port connections (_.name_) than if using explicit connection (_.name (expression)_). 

In the example below, the leading dot in  _.Clk_ gets a different color than the leading dot in _.A(A)_:

```
module Test ();
    Submodule S (
        .Clk,
        .A(A)
    );
endmodule
```

TheClams, 2023-01-08:
Both have same color for me, it must be related to the color scheme you are using: both use the scope `support.function.port` , but there is an extra indicator for .Clk which is `support.function.port.implicit` : if your color scheme has specific color for this scope this might explain the difference.

maxbjurling, 2023-01-13:
Thanks for looking at this. I removed the SystemVerilog package and then installed it again and now the syntax highlighting looks OK for the example above (using default color scheme Monokai). So something must have been corrupted in my package/installation/settings etc.