TheClams/SystemVerilog #62: Alignment of logic and wire on input/output ports

code-tangent, 2023-02-09:
Not sure if there is something I need to change in my config file but when I try to run Verilog:Alignment I get a strange alignment with my input wires and output logic ports. 

Im using version 3.1.10 of the package and build 4126 of sublime.

[image]

TheClams, 2023-02-09:
This is the expected behavior: the alignment is done to align each part of a port declaration together: direction, net type, data type (vector type + signing + vector width, or non vector type), name and unpacked dimensions.
Here there is a mix of net type (e.g. wire or var) and data type (e.g. logic, int, ...), and that's why they are aligned in different "column"

code-tangent, 2023-02-09:
Thank you for the fast response!

closed 2023-02-09 by code-tangent